TF-RMM/tf-rmm #5: pin submodules to commits

flit, 2023-06-02:
The git submodules need to be pinned to specific commits.

Because of this, the `TF-RMM Release v0.3.0` tag no longer builds due to an out of date reference to a t_cose source file in [libs/t_cose/CMakeLists.txt](https://github.com/TF-RMM/tf-rmm/blob/tf-rmm-v0.3.0/lib/t_cose/CMakeLists.txt).

For reference, the error is:
```
CMake Error at lib/t_cose/CMakeLists.txt:6 (add_library):
  Cannot find source file:

    /work/build/clones/tf-rmm/ext/t_cose/crypto_adapters/t_cose_mbedtls_crypto.c
```

soby-mathew, 2023-06-02:
The submodules are pinned to the required commit for the dependencies. Please do `git submodule sync` and `git submodule update` from RMM root directory.

`git submodule status` should show the below out for tf-rmm-v0.3.0 tag

```
 67d2dfd41e13f09ff218aa08e2d35f1c32f032a1 ext/cpputest (v4.0)
 869298bffeea13b205343361b7a7daf2b210e33d ext/mbedtls (v3.2.1)
 7132236977a2f8f54a5f80f389ad5e45de73a49e ext/qcbor (v1.0~4)
 cecf62bf4275eae732fadb067dc9106c7815ced4 ext/t_cose (remotes/origin/add-restartable-api)
```

Perhaps there are other ways to pin submodules, please let us know if you think this can be done in a better way.

flit, 2023-06-06:
You are entirely correct, it was a configuration related issue on my end. My apologies! I'm used to having git configured to automatically sync submodules, or use a GUI git interface that does this.

closed 2023-06-06 by flit

soby-mathew, 2025-01-13:
> 
[fix(doc): bump jinja2 version from 3.1.4 to 3.1.5](https://github.com/TF-RMM/tf-rmm/commit/cbabd841df9b402f712be982554372b1c80c2359) 

The commit message referencing this issue was wrong. It closed the dependabot issue:  https://github.com/TF-RMM/tf-rmm/security/dependabot/5

commit cbabd841 (2025-01-13): fix(doc): bump jinja2 version from 3.1.4 to 3.1.5

This patch upgrades Jinja2 from version 3.1.4 to version 3.1.5
as per dependabot suggestion.

Closes #3
Closes #5

Change-Id: Ib8d19babe913e6d40fca6f4336c767e999ea94fe