TF-RMM/tf-rmm #18: Number of return values in rmm_handler does not match with EL3(TF-A)

Yiwei-Zhuang, 2024-03-13:
In RMM, [rmm_handler](https://github.com/TF-RMM/tf-rmm/blob/main/runtime/core/aarch64/entry.S) returns X<sub>0</sub>-X<sub>5</sub> to NS host. However, I notice in TF-A, [rmmd_rmi_handler](https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/tree/services/std_svc/rmmd/rmmd_main.c) overwrites the X<sub>5</sub> register which is the forth output of a RMI request when the case is `RMM_RMI_REQ_COMPLETE`. I tested `smc_rtt_read_entry` and found that the value of RIPAS is always 0.

RMM `smc_rtt_read_entry`                           
x0  SMC_RMM_REQ_COMPLETE
x1  RMI_SUCCESS
x2  last_level
x3  HIPAS
x4  pa
x5  RIPAS

AlexeiFedorov, 2024-03-14:
I modified smc_rtt_read_entry() to return 0xAA55 instead of RIPAS_EMPTY (0) and added logging of ret_val.ret0-ret4 in TFTF's  host_rmi_handler(). This the output I got:
RMM:
SMC_RMM_RTT_READ_ENTRY        88219000 0 0 > RMI_SUCCESS 0 2 8821c000 aa55
TFTF:
> Executing 'Realm EL1 creation and execution test'
INFO:    Realm start adr=0x88119000
[VMID 0][Rec 0]: going to sleep for 20ms
INFO:    === 0x0
INFO:    === 0x0
INFO:    === 0x2
INFO:    === 0x8821c000
INFO:    === 0xaa55

AlexeiFedorov, 2024-03-14:
rmmd_rmi_handler(..):
..
	switch (smc_fid) {
	case RMM_RMI_REQ_COMPLETE: {
		uint64_t x5 = SMC_GET_GP(handle, CTX_GPREG_X5);

		return rmmd_smc_forward(REALM, NON_SECURE, x1,
					x2, x3, x4, x5, handle);
	}

Yiwei-Zhuang, 2024-03-14:
Oh, It is my mistake. It seems my local TF-A code misses this [patch](https://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/commit/?id=ea7aee20c1fed0d206020ccb8541c2fa8d4a3795) which looks like
```
 	switch (smc_fid) {
	case RMM_RMI_REQ_COMPLETE:
		return rmmd_smc_forward(REALM, NON_SECURE, x1,
					x2, x3, x4, 0, handle);
```
The RMI request can return the fourth argument correctly now.

closed 2024-03-14 by Yiwei-Zhuang