TF-RMM/tf-rmm #19: Why is Realm VM running so slow?

bows7ring, 2024-03-21:
Hi there! 
I've successfully built the RMM enviroments according to the [tutorial](https://linaro.atlassian.net/wiki/spaces/QEMU/pages/29051027459/Building+an+RME+stack+for+QEMU#QEMU-VMM) written by @jpbrucker and  Alex Bennée. 
(I know this is not the best place to ask, but I don't have permission to comment in that turotial..)

It took me 20 mins to boot the guest Realm VM, which makes it hard to test and debug.
I'm curious if it's normal for Realm to run this SLOW?
Is It because of RMM or Qemu? How can I speed it up?
 (I've already switched the Firmware to the Release version, but it seems nothing has changed).

bows7ring, 2024-03-24:
Thanks, ShrinkWrap helps a lot. Testing RMM is much more convenient now.

But my development work still relies heavily on Qemu( both host simulation and guest VMM). Do you have any suggestions for using Qemu in FVP? (If not, feel free to close the issue.)

Thank you for your reply！

ryanhrob, 2024-03-25:
Hi, There is a thread here ([https://gitlab.arm.com/tooling/shrinkwrap/-/issues/13](https://gitlab.arm.com/tooling/shrinkwrap/-/issues/13)) about using QEMU as the guest VMM under Shrinkwrap. Although I must admit I haven't been following it so not sure if it exactly aligns with what you are trying to do.

bows7ring, 2024-03-26:
Wow, thanks, this is exactly what I am working on.
But I still have some problems running Qemu(my qemu vmm has no output), maybe I will continue seeking for help in the shrinkwrap repo.

closed 2024-03-26 by bows7ring

Yiwei-Zhuang, 2024-06-18:
Have you ever met the situation that the arch_timer frequently make realm vm exit? It causes lots of world change between realm EL0 to NS EL2 since all irqs are handled by kvm. Will this behavior significantly slow down the realm vm? Compared with the vm in NS, with same simulator, realm vm performs much worse than normal vm when they only did calculating tasks so the memory should not matter. We are trying to figure out the reason so I count the rec_exit reason and notice nearly all of them are caused by arch_timer interrupt.

jpbrucker, 2024-06-18:
That's a very interesting observation. Have you tried reducing the timer frequency in the guest, for example CONFIG_HZ_100? It definitely feels faster in my tests

Yiwei-Zhuang, 2024-06-19:
Yes, I tried it. Although the simulator could not show the real performance, but the kernel with CONFIG_HZ_100 looks better than the CONFIG_HZ_250 kernel. So, the world change is truly a bottleneck, right? Will there be any plan to optimize such long interrupt path? The normal vm does not need a world change and both kernel and hypervisor(kvm) are running on same exception level.

soby-mathew, 2024-06-19:
Potentially there is an optimization that can be done for this case in RMM but that would be complex as I understand. It was assumed that timer interrupts don't happen often enough to be a real problem. RMM also needs to ensure that there aren't any other pending interrupts in this optimized flow path. There are solutions but this is not something we have planned to do.