TF-RMM/tf-rmm #26: How to use rsi_attestation_token_continue

Bercoke, 2024-05-19:
The memory allocated using kmalloc in the directories linux/arch/arm64/kvm and linux/arch/arm64/kernel fails the memory check performed by rsi_attestation_token_continue.

Memory detection is shown in the figure
[image]

I would like to know where to call the memory allocation function to obtain memory addresses that can pass the memory check.

suzukikp, 2024-05-20:
Please could you share the calls made by your Linux code ? Are you passing the "PA" of the memory to the RSI call ? 
We have hooked the attestation calls to the Generic COCO TSM CONFIG framework. The patch could be found here :

https://gitlab.arm.com/linux-arm/linux-cca/-/commit/7e5aabb8d7d03f5afcb7224b892eb646ccea91df#7eda40b94a7a616d2ddfd3e218f2632a9e023163_0_76

soby-mathew, 2024-07-09:
Please feel to reopen if you still have queries on this issue.

closed 2024-07-09 by soby-mathew