TF-RMM/tf-rmm #28: Virtualization Host Extensions (VHE) support in tf-rmm

XiangyiXuWashU, 2024-06-21:
We have a question about Virtualization Host Extensions (VHE), which force the host kernel to run in EL2 to reduce virtualization overhead. However, RMM is very similar to a Type 1 hypervisor, and RMM is already running in R-EL2. In your implementation, why do you enable VHE by setting HCR_EL2.E2H? What is the benefit of that? Meanwhile, we found that Hafnium also enables the VHE feature in S-EL2. We are considering enabling the VHE feature in virtCCA, but we are skeptical about whether VHE will improve performance.

soby-mathew, 2024-06-21:
Hi @XiangyiXuWashU , 

For RMM, the VHE is enabled mainly for the ability to split the Virtual Address space to High and Low VA spaces. This allows RMM to use the High VA for per CPU private mapping and the Low VA for the global mappings. The private mapping allows RMM to isolate any mappings specific to the Realm to the particular CPU. For more details on how the 2 VA spaces are used, refer here : https://tf-rmm.readthedocs.io/en/latest/design/memory-management.html#rmm-stage-1-translation-regime

This design gives better security properties. Also we have some plans to utilize EL0 of the EL2&0 translation regime to run some parts of RMM at a lower privilege level. So we are seeing this as a capability to enhance security and isolation properties in RMM. I am not sure whether enabling E2H gives any additional performance.

soby-mathew, 2024-07-09:
Please feel free to reopen if you have more questions on this topic.

closed 2024-07-09 by soby-mathew