TF-RMM/tf-rmm #29: Live Migration: How to Sync Realm's Timer

bows7ring, 2024-07-08:
Currently, migration can only succeed when VM is at very begining of Kernel boot. In this scenario, guest kernel have not set its timer, `cntv_cval` should be 0, and guest can then run successfully on dst host with a newly set timer.

- If we migrate the VM when it enter the guest kernel shell, it will encounter a guest kernel panic at `cpu_idle` , like this:
[image]
 We name it as "Zombie VCPU", because Realm can still enter and exit REC, but REC's timer info doesn't change anymore, unlike normal ones.

- Technically, arch_cpu_idle is nothing but a `wfi` , so the kernel crashed because it is waiting for a timer interrupt that never come.
- If migrate the VM right before guest kernel enter its shell, it will crash as well, but the guest kernel's panic info will be useless.

Based on a series of tests, we can confirm that this is caused by virtual timer interruption.

---

According to my understanding, Realm's timer virtual interrupt works like this:
[image]

I tried logging timer info in this flow. And I think it is **Step 2**, the vGIC physical interrupt that is not working.

---

Acording to RMM spec and RMM code:
```c
void report_timer_state_to_ns(struct rmi_rec_exit *rec_exit)
{
	/* Expose Realm EL1 timer state */
	rec_exit->cntv_ctl = read_cntv_ctl_el02();
	rec_exit->cntv_cval = read_cntv_cval_el02() - read_cntvoff_el2();

	rec_exit->cntp_ctl = read_cntp_ctl_el02();
	rec_exit->cntp_cval = read_cntp_cval_el02() - read_cntpoff_el2();
}
```
[image]

We can see that `cntvoff` is fixed at 0, physical timer is disabled in Realm, and Realm use the physical counter value as virtual counter.

If we want to sync the timer after migration, how should we set the REC's timer?
The easiest way would be to set a `cntvoff = cntpct_el0(src) - cntpct_el0(dst)` after migration?

> On src host:
>   cntvoff1 = 0
>   vCount1 = pCount1 - cntvoff = pCount1
> 
> On dst host:
>   vCount2 = vCount1 = pCount1
>   vCount2 = pCount2 - cntvoff2
> 
> ∴ cntvoff2 = pCount2 - pCount1

Is that logically correct? We're trying to start here:
[image]

soby-mathew, 2024-08-02:
Hi @bows7ring , with my limited understanding on how kernel expects time, I think the main thing to ensure is that the migrated VM does not see the time go backwards. It might be worth printing the values of the EL1 views of the timers registers and ensure that this is the case.

It seems that the timer is not initialized properly in the restored VM and hence timer may not be counting/generating interrupt. Might be worth ensuring the values of the timer control registers in the restored VM.

bows7ring, 2025-02-26:
This was caused by PAC. Pointer checking kills the migrated kernel at a very early place, makes it looks like a vtimer-bug. We haven't implemented PAC migrationt yet, just bypassed here.

closed 2025-02-26 by bows7ring

soby-mathew, 2025-02-26:
Ack, thanks for the update on this.