TF-RMM/tf-rmm #31: How to Enable Debugging for Realm VMs in the Arm CCA Framework

Crescentm, 2024-08-01:
I know this sounds unreasonable, especially since the gdb debugging of Realm VMs is specifically disabled in the Host Linux KVM module, which invalidates the debugging annotations of qemu vmm. However, I do have a corresponding debugging need. I need to perform gdb debugging on the Guest Linux kernel to identify the problem. 

**Is there any corresponding method, or when can debugging be implemented through RMM**? Thank you very much.

soby-mathew, 2024-08-01:
There are 2 debug models possible for Realm Guest , one is the **Self hosted debug** wherein a debugger running in the Realm Guest Kernel debugs software within the same VM and the other is **Host Debug of Realm** wherein the debugger is running on the Host Linux side and debugs the target Realm Guest VM.

Self hosted debug is specified in the RMMv1.0 specification but TF-RMM doesn't implement the support yet. We hope to add this support in the following months.

Host debug of Realm is not supported in RMMv1.0 but it will be specified as part of RMMv1.1. Hence this will have to wait till the specification is ready in future.

Crescentm, 2024-08-01:
Thank you, this helped me a lot.

closed 2024-08-01 by Crescentm