TF-RMM/tf-rmm #32: plat/arm/arm_memory: Cannot get correct granule addr from idx when num_bank=1

yanghaku, 2024-08-07:
In function ```arm_set_dram_layout```, when ```num_bank``` is ```1```, the variable ```idx_bank_1``` will be ```0```.
In this case, the function ```plat_granule_idx_to_addr``` will return the wrong address:
```c++
unsigned long plat_granule_idx_to_addr(unsigned long idx)
{
	assert(idx < arm_dram.num_granules);

	if (idx < arm_dram.idx_bank_1) {     //  The code cannot go to this branch!
		return arm_dram.arm_bank[0].start_addr + (idx * GRANULE_SIZE); 
	}

	return arm_dram.arm_bank[1].start_addr +
			((idx - arm_dram.idx_bank_1) * GRANULE_SIZE);
}
```

So I suggest to add this patch:
```diff
diff --git a/plat/arm/src/arm_memory.c b/plat/arm/src/arm_memory.c
index 01783df..ba424f8 100644
--- a/plat/arm/src/arm_memory.c
+++ b/plat/arm/src/arm_memory.c
@@ -41,6 +41,9 @@ void arm_set_dram_layout(struct ns_dram_info *plat_dram)
 
 		bank_ptr++;
 	}
+	if (num_banks < MAX_DRAM_NUM_BANKS) {
+		dram_ptr->idx_bank_1 = num_granules;
+	}
 
 	dram_ptr->num_granules = num_granules;
 

```

soby-mathew, 2024-09-02:
Indeed, thank you @yanghaku for the bug report.

Could you contribute a patch for the same ?

soby-mathew, 2024-09-03:
Hi @yanghaku, 

There is this refactor to add multiple DDR banks here : https://review.trustedfirmware.org/c/TF-RMM/tf-rmm/+/27372/35/plat/arm/src/arm_memory.c 

I think this should resolve the issue, could you please check and let us know ? 

Thanks.

yanghaku, 2024-09-06:
@soby-mathew  OK, I have tested it and it works with 1 DRAM bank on the QEMU platform.

Besides, I think there is no need to use the binary search for idx/addr translation when the platform has 1 or 2 DRAM banks. Maybe the user can choose the algorithm when building or the RMM can automatically choose in the boot stage.

Thanks.

closed 2024-09-06 by yanghaku

soby-mathew, 2024-09-06:
Thanks for feedback. 

Yes, that could be a possible future optimization. We would need some performance measurement to justify that binary search for 1 or 2 DDR banks has performance impact.
Another alternative is that the platform could provide its own implementation to compute the granule idx.