TF-RMM/tf-rmm #38: Will RMM try to cache slot_buffers of RD, REC, REC_AUX...?

Yiwei-Zhuang, 2025-02-24:
Every RMI_REC_ENTER maps rd, rec and rec_aux to transient table and unmap after use. I have not tried yet but would not those dsb(), isb() operations slow down a lot in Realm world? If so, will there be any optimization on it like cache them? Thanks for any explanation!

soby-mathew, 2025-02-24:
The REC and REC_AUX are mapped while the realm is executing as Realm can issue RSI cmds which would need information from these. Note that these buffer are mapped to private VA space of the CPU and hence the dsb is Non-SH and hence expected to be lightweight. I need to check further, but it does seems the isb() may not be needed. 

The RD is mapped for a short duration to get the `realm_state` and then unmapped. 

Mapping these on  a need-basis is good for security and also avoids speculative side-channels. Hence we have the current design.

Yiwei-Zhuang, 2025-02-24:
Thanks for your information.

Depending on the rmm's implement, during RMI_REC_ENTER, `buffer_aux_granules_map` does `num_aux`(max is 16) times dsb and isb separately. Also, I noticed `GRANULE_STATE_PDEV_AUX` and `GRANULE_STATE_VDEV_AUX` are added. I do not know whether those granules will be implemented with same logic(like in a loop with several barriers),  but It looks a big waste to me.

Thus, what if we map rec, rec_aux for each realm instance when we do RMI_CREATE_REC? And may be we can use a simple hash table to find out the correct va to directly read/write. Will there be side-channels risks?

From my understanding, since only entries' value is modified, we only need to ensure the new data is visible to others. If the isb() is removed, it seems mapping while needed should not take too much time.

Glad to hear further progress if there are improvements during slot_buffer mapping.

soby-mathew, 2025-02-25:
Hi Yiwei-Zhuang,

> num_aux(max is 16) times dsb and isb separately. Also, I noticed GRANULE_STATE_PDEV_AUX and GRANULE_STATE_VDEV_AUX are added. I do not know whether those granules will be implemented with same logic(like in a loop with several barriers), but It looks a big waste to me

You are right, the current aux granule mapping logic is not efficient. There are 2 optimizations that are possible :

1. Introduce an API to map several aux granules in one go.  This means that the isb() and dsb() can be issued just once for the multiple granules.
2. In most of the cases, we dont need to map all the 16 aux granules. It would be possible to map just the needed aux granule. 

These 2 are good optimizations that can be done on the code base. I can raise 2 issues in github for doing this. It would be helpful if you can contribute changes for this.

> we map rec, rec_aux for each realm instance when we do RMI_CREATE_REC? And may be we can use a simple hash table to find out the correct va to directly read/write. Will there be side-channels risks?

hmm, there could 1000's of REC at runtime, and RMM cannot allocated Stage 1 translation tables to accommodate all the mapping all the REC's. Plus having all the REC and AUX granules mapped in RMM is a security hazard as it increases the mapped memory foot print of RMM. In addition to speculative access to the mapped memory, any vulnerability in RMM can be used to access/modify the mapped contents. Hence we try to keep the mapped memory minimal in RMM.

>  If the isb() is removed, it seems mapping while needed should not take too much time.

I checked with Architecture folks in Arm. So mapping a new Stage 1 RW, non-executable, cacheable memory does require an isb() before RMM makes an access to the new page. If FEAT_ETS2 is present, then the isb() can be avoided. FEAT_ETS2 is not mandatory in Arm 9.2 so we cannot remove the isb() unconditionally from RMM code.

Yiwei-Zhuang, 2025-02-27:
I am glad to contribute on the issue of optimizing mapping several granules one time.

For the isb instruction, I checked the latest Arm Architecture Reference Manual, in D24.2.83 about ID_AA64MMFR1_EL1 and it says "From Armv8.8, the values 0b0000 and 0b0001 are not permitted". Two codes 0b0000 and 0b0001 stand for "Enhanced Translation Synchronization is not supported." The code left is 0b0010 which stands for "FEAT_ETS2 is implemented." Does not that mean the FEAT_ETS2 is mandatory from ARM v8.8?

soby-mathew, 2025-02-27:
Thanks for your help, I have raised a github issue : https://github.com/TF-RMM/tf-rmm/issues/39

> Does not that mean the FEAT_ETS2 is mandatory from ARM v8.8?

Indeed, FEAT_ETS2 is mandatory from ARM v8.8 which means it is mandatory from ARM v9.3. FEAT_RME is from ARM v9.2 and hence RMM cannot unconditionally depend on 9.3 extension. Potentially we could have a check for FEAT_ETS2 and skip the isb() in RMM code.

This is a screen shot from latest Arm ARM showing FEAT_ETS2 :

[image]

closed 2025-03-05 by Yiwei-Zhuang